AltBeacon/android-beacon-library #1200: Notifications are not displayed on Android 14 in the background

jirka-oone, 2024-08-23:
### Expected behavior
notification is triggered

### Actual behavior
notification not triggered when beacon is found

### Steps to reproduce this behavior
I ran a test app [https://github.com/davidgyoung/android-beacon-library-reference-kotlin] - scanning beacon Eddystone. The app works fine in the foreground, the notification was displayed that it found the beacon in the nearby, but when I tried a similar steps in the background, the notification was not displayed. test on android 12, 13 was successful and without problems, 
do you know what could be wrong, thx

### Mobile device model and OS version
test device: galaxy a14, android 14

### Android Beacon Library version
8cd351f093bad282a03ef452c1baf4650aaff246
2.0-beta.6-624-g8cd351f

davidgyoung, 2024-11-15:
Did you really use the library version from this commit?  https://github.com/AltBeacon/android-beacon-library/commit/8cd351f093bad282a03ef452c1baf4650aaff246

That library code is from 7 years before Android 14 was released so there is no way it will work to detect beacons in the background on Android 14.  You must use a much newer library version.  The latest is [2.21.0-beta3](https://mvnrepository.com/artifact/org.altbeacon/android-beacon-library/2.21.0-beta3) which has specific fixes for Android 14.

davidgyoung, 2024-11-16:
This may have been fixed by #1205 or #1208

Buthrakaur, 2025-02-11:
Hello @davidgyoung ,
what is the expected beacon detection delay when the app is in the background, please? We mostly see the delay around 5 minutes which is too long for our use-case - we'd need to get the delay way down to e.g. 5 seconds. Is that achievable?

davidgyoung, 2025-03-16:
@Buthrakaur  see this page for expected detection times, which can be brought down to 5 seconds with a number of different configuration options

https://altbeacon.github.io/android-beacon-library/detection_times.html

closed 2025-03-16 by davidgyoung