AltBeacon/android-beacon-library #1201: ForegroundServiceStartNotAllowedException Crash Despite Fallback Mechanism in v2.20.6

pubudithajayasekara, 2024-09-11:
### Issue Description

We are encountering a `ForegroundServiceStartNotAllowedException` in our application when using the `org.altbeacon:android-beacon-library:2.20.6`. According to the documentation, starting with library version 2.19.5-beta6, the library should handle this exception by falling back to using the Job Scheduler for scans when starting a foreground service is not allowed by the operating system. However, we are still experiencing crashes related to this issue.

### Stack Trace
ForegroundServiceStartNotAllowedException: startForegroundService() not allowed due to mAllowStartForeground false ...

### Expected Behavior

According to the documentation, the library should catch the `ForegroundServiceStartNotAllowedException` and use the Job Scheduler as a fallback, resuming the use of a foreground service when the app returns to the foreground.

### Actual Behavior

Despite using version 2.20.6, the application is crashing due to the `ForegroundServiceStartNotAllowedException`, which suggests that the fallback mechanism to the Job Scheduler may not be working as intended.

### Steps to Reproduce

1. Integrate `org.altbeacon:android-beacon-library:2.20.6` into the application.
2. Attempt to start a foreground service while the app is in the background.
3. Observe the crash related to `ForegroundServiceStartNotAllowedException`.

### Additional Information

- **Android Version**: 12+ 
- **Device Models**: [Cricket U680AC , google Pixel 8 , motorola , OPPO CPH2385 , samsung SM-A045F, Nokia Nokia G11 Plus ....]
- **Library Version**: `org.altbeacon:android-beacon-library:2.20.6`

### Documentation Reference

- [Foreground Service Documentation](https://altbeacon.github.io/android-beacon-library/foreground-service.html)

slai47, 2024-10-14:
Bumping this as this error is still occurring on 2.20.7.

Logs from Crashlytics from our production:

```Fatal Exception: java.lang.RuntimeException: Unable to create service org.altbeacon.beacon.service.BeaconService: android.app.ForegroundServiceStartNotAllowedException: Service.startForeground() not allowed due to mAllowStartForeground false: service com.adt.adtplus.prod/org.altbeacon.beacon.service.BeaconService
       at android.app.ActivityThread.handleCreateService(ActivityThread.java:4925)
       at android.app.ActivityThread.-$$Nest$mhandleCreateService()
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2407)
       at android.os.Handler.dispatchMessage(Handler.java:107)
       at android.os.Looper.loopOnce(Looper.java:232)
       at android.os.Looper.loop(Looper.java:317)
       at android.app.ActivityThread.main(ActivityThread.java:8592)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)```
       
       ```          Caused by android.app.ForegroundServiceStartNotAllowedException: Service.startForeground() not allowed due to mAllowStartForeground false: service com.adt.adtplus.prod/org.altbeacon.beacon.service.BeaconService
       at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:54)
       at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:50)
       at android.os.Parcel.readParcelableInternal(Parcel.java:5016)
       at android.os.Parcel.readParcelable(Parcel.java:4998)
       at android.os.Parcel.createExceptionOrNull(Parcel.java:3178)
       at android.os.Parcel.createException(Parcel.java:3167)
       at android.os.Parcel.readException(Parcel.java:3150)
       at android.os.Parcel.readException(Parcel.java:3092)
       at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6960)
       at android.app.Service.startForeground(Service.java:776)
       at org.altbeacon.beacon.service.BeaconService.startForegroundIfConfigured(BeaconService.java:279)
       at org.altbeacon.beacon.service.BeaconService.onCreate(BeaconService.java:201)
       at android.app.ActivityThread.handleCreateService(ActivityThread.java:4912)
       at android.app.ActivityThread.-$$Nest$mhandleCreateService()
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2407)
       at android.os.Handler.dispatchMessage(Handler.java:107)
       at android.os.Looper.loopOnce(Looper.java:232)
       at android.os.Looper.loop(Looper.java:317)
       at android.app.ActivityThread.main(ActivityThread.java:8592)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)
        ```
        
        Android Version 14 is only effected for us. Replication steps are the same as above. 
        
        Devices are Samsung and Google. S24, 23 Ultra, Pixel 8, 7 and 6 for examples. 
        
        Version 2.20.6 on production, 2.20.7 on staging. Both having similar issues.

slai47, 2024-10-14:
Could this be related to https://issuetracker.google.com/issues/307329994 and needing to add in more permission settings with ServiceCompat?

oebb, 2024-11-06:
We're also having the same issue w Android 14 devices (targeting Android 14)

davidgyoung, 2025-03-15:
Cannot reproduce using the official reference app for Kotlin which targets Android 14. (targetSdkVersion 34).  These tests were performed with library version 2.21.0, but I have no reason to think that the incremented library version makes the difference.

Here is a [branch of the Kotlin reference app](https://github.com/davidgyoung/android-beacon-library-reference-kotlin/tree/foreground-service-fallback-test) that I created to try to reproduce the problem.  It does this:

1. Starts ranging and monitoring after app start up and not using a foreground service.
2. Waits 30 seconds so the user can manually put the app to the background.
3. Tells the library to switch to a foreground service (now that the app is in the background.)

Logging output is this:

```
2025-03-15 13:21:04.890 15893-15893 BeaconManager           org.altbeacon.beaconreference        I  Attempting to starting foreground beacon scanning service.
2025-03-15 13:21:04.897 15893-15893 BeaconManager           org.altbeacon.beaconreference        W  Foreground service blocked by ServiceStartNotAllowedException.  Falling back to job scheduler
```

No crash is observed.

I have no doubt that there is a problem here given the Crashlytics reports.  But we need to find a way to reproduce the issue with the reference app  outside of reports from Crashlytics.  If the only reports of problems are from Crashlytics, then the cause of the problem may be the configuration of the library in a custom app.  We must eliminate that possibility by reproducing with the official reference app in order to confirm a library problem and provide a fix.

@oebb , @slai47 , @pubudithajayasekara can you please try the branch above and see if you can reproduce the problem?  If not, can you perhaps put similar changes into your own apps and see if you can reproduce the problem?  If you can reproduce the problem with your own app but not the reference app, you may be able to look for differences in the way beacon scanning is set up to find a solution to the problem.

I am going to close this issue until we have a reproducible test case with the reference app, but I encourage you to try this and re-open if you can reproduce with the reference app.

closed 2025-03-15 by davidgyoung

slai47, 2025-03-27:
For anyone who runs into this issue, my team found the problem and it has to do with WHICH context you start the BeaconManager with. For my team we started the BeaconManager on app create with the application context but in order for it to work in the background, you will need to create it with the context from the receiver you have. This meets the requirements Android puts on background tasks and will allow for AltBeacon to work as it should without a crash. Hope this helps others!