AltBeacon/android-beacon-library #1203: Beacon scanning stops after 10 minutes when app is on foreground and enableForegroundServiceScanning is used.

elican-doenyas, 2024-09-27:
### Expected behavior
Beacon scanning should continue as usual independent of the time app is used.

### Actual behavior
Beacon scanning stops after ~10 minutes while app is on foreground and enableForegroundServiceScanning is called. This is also reproducible in the AltBeaconReference application.

### Steps to reproduce this behavior
- Call beaconManager.enableForegroundServiceScanning
- Start ranging beacons with call to beaconManager.startRangingBeacons(<region>)
- Open the app and leave it on foreground for 10+ minutes
- Observe that beacon scanning halts after ~10 minutes

### Mobile device model and OS version
Samsung S23 - Android 14

### Android Beacon Library version
org.altbeacon:android-beacon-library:2.20.5 & org.altbeacon:android-beacon-library:2.20

elican-doenyas, 2024-09-27:
@davidgyoung I have pinpointed the reason for this issue, the DEFAULT_SCAN_TIMOUT_MILLIS was reduced from 30 minutes to 10 minutes in the Android OS with the following commit;

https://android-review.googlesource.com/c/platform/packages/modules/Bluetooth/+/2237636/1/android/app/src/com/android/bluetooth/btservice/AdapterService.java 

So the variable in the CycledLeScanner class to restart scanning and enforce long scans below;

    public static final long ANDROID_N_MAX_SCAN_DURATION_MILLIS = 30 * 60 * 1000l; // 30 minutes

 should be switched to 10 minutes.

Also, as far as I understand, in order to enforce long scans we need to set the mLongScanForcingEnabled parameter of CycledLeScanner, but I could not find where we can set this as an integrator of the library.

davidgyoung, 2024-10-21:
@elican-doenyas  apologies for the delayed response here.

The limitation in long-running scans has been around since Android 8.1, but I was unaware that they reduced it from 30 min to 10 min until you reported this issue.  The good news is that the library already has a built-in workaround for this as described in #529.  The bad news is that the library mechanism to work around this limitation needs to be adjusted to 10 min.

I just made a new  change that updates the library's interval for long scans from 30 mins to 10 mins match the change in Android https://github.com/AltBeacon/android-beacon-library/pull/1207

In order to make this work, you have to update to 2.21.0-beta2 (just published today, it may take a few hours to sync) and then enable this option in your manifest as descried in #529

**OR** easier yet, library version 2.21 and higher  has a new settings api that makes this much easier to configure:

```
        // Use new configuration API in library version 2.21
        val settings = Settings(scanStrategy = Settings.IntentScanStrategy(), longScanForcingEnabled = true)
        // This line below will apply the new beacon scanning settings immediately  Any individual
        // settings not specified on the new settings object will revert to defaults.
        // If you only want to make a partial change to the settings, without reverting to defaults
        // for any settings unspecified you may call `beaconManager.adjustSettings(settings)`
        beaconManager.replaceSettings(settings)
```

davidgyoung, 2024-10-21:
Fixed in #1207

PR #1207 (2024-11-16): Reduce max scan interval from 30 min to 10 min to fix droputs
fixes #1203
closed 2024-10-21 by davidgyoung