AltBeacon/android-beacon-library #1206: java.lang.IllegalStateException: schedule()/enqueue() called more than 250 times in the past 60000ms

Muran-Hu, 2024-10-14:
### Expected behavior

Do not get `IllegalStateException: schedule()/enqueue() called more than 250 times in the past 60000ms` anymore.

### Actual behavior

The app will crash if we call `org.altbeacon.beacon.service.ScanJobScheduler.schedule()` more frequently.

```
0-com.example.xxx has called schedule() too many times
Shutting down VM
FATAL EXCEPTION: main
Process: com.example.xxx, PID: 8115
java.lang.RuntimeException: Unable to pause activity {com.example.xxx/com.example.xxx.demoapp.MainActivity}: java.lang.IllegalStateException: schedule()/enqueue() called more than 250 times in the past 60000ms. See the documentation for more information.
	at android.app.ActivityThread.performPauseActivityIfNeeded(ActivityThread.java:6454)
	at android.app.ActivityThread.performPauseActivity(ActivityThread.java:6403)
	at android.app.ActivityThread.handlePauseActivity(ActivityThread.java:6335)
	at android.app.servertransaction.PauseActivityItem.execute(PauseActivityItem.java:56)
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:50)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:190)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:105)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:3082)
	at android.os.Handler.dispatchMessage(Handler.java:117)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:293)
	at android.app.ActivityThread.loopProcess(ActivityThread.java:9986)
	at android.app.ActivityThread.main(ActivityThread.java:9975)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:586)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1240)
Caused by: java.lang.IllegalStateException: schedule()/enqueue() called more than 250 times in the past 60000ms. See the documentation for more information.
	at android.os.Parcel.createExceptionOrNull(Parcel.java:2451)
	at android.os.Parcel.createException(Parcel.java:2427)
	at android.os.Parcel.readException(Parcel.java:2410)
	at android.os.Parcel.readException(Parcel.java:2352)
	at android.app.job.IJobScheduler$Stub$Proxy.schedule(IJobScheduler.java:312)
	at android.app.JobSchedulerImpl.schedule(JobSchedulerImpl.java:47)
	at org.altbeacon.beacon.service.ScanJobScheduler.schedule(ScanJobScheduler.java:235)
	at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:96)
	at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:103)
	at org.altbeacon.beacon.BeaconManager.applyChangesToServices(BeaconManager.java:1370)
	at org.altbeacon.beacon.BeaconManager.updateScanPeriods(BeaconManager.java:1354)
	at org.altbeacon.beacon.BeaconManager.setBackgroundModeInternal(BeaconManager.java:683)
	at org.altbeacon.beacon.BeaconManager.setBackgroundMode(BeaconManager.java:659)
	at org.altbeacon.beacon.powersave.BackgroundPowerSaverInternal.onActivityPaused(BackgroundPowerSaverInternal.java:80)
	at android.app.Application.dispatchActivityPaused(Application.java:453)
	at android.app.Activity.dispatchActivityPaused(Activity.java:1506)
	at android.app.Activity.onPause(Activity.java:2521)
	at androidx.fragment.app.FragmentActivity.onPause(FragmentActivity.java:282)
	at android.app.Activity.performPause(Activity.java:8786)
	at android.app.Instrumentation.callActivityOnPause(Instrumentation.java:1553)
	at android.app.ActivityThread.performPauseActivityIfNeeded(ActivityThread.java:6444)
	at android.app.ActivityThread.performPauseActivity(ActivityThread.java:6403) 
	at android.app.ActivityThread.handlePauseActivity(ActivityThread.java:6335) 
	at android.app.servertransaction.PauseActivityItem.execute(PauseActivityItem.java:56) 
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:50) 
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:190) 
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:105) 
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:3082) 
	at android.os.Handler.dispatchMessage(Handler.java:117) 
	at android.os.Looper.loopOnce(Looper.java:205) 
	at android.os.Looper.loop(Looper.java:293) 
	at android.app.ActivityThread.loopProcess(ActivityThread.java:9986) 
	at android.app.ActivityThread.main(ActivityThread.java:9975) 
	at java.lang.reflect.Method.invoke(Native Method) 
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:586) 
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1240) 
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.job.JobSchedulerService.scheduleAsPackage(JobSchedulerService.java:1141)
	at com.android.server.job.JobSchedulerService.scheduleAsPackage(JobSchedulerService.java:1097)
	at com.android.server.job.JobSchedulerService$JobSchedulerStub.schedule(JobSchedulerService.java:2983)
	at android.app.job.IJobScheduler$Stub.onTransact(IJobScheduler.java:155)
	at android.os.Binder.execTransactInternal(Binder.java:1192)
callee: null 1840/4858
```

### Steps to reproduce this behavior

1. We have two activities A and B, calling `org.altbeacon.beacon.service.ScanJobScheduler.schedule()` method on the lifecycle method `onStart()` of Activity B.
2. Start Activity B from A, back to Activity A
3. Repeat step 2 several times
4. App crash and report the above errors

### Mobile device model and OS version

1. HUAWEI P50 Pro
2. OS: HarmonyOS version 4.2.0

### Android Beacon Library version

2.20.6

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

Muran-Hu, 2024-10-14:
Is there anyone who could help look into this issue? 

I'm wondering if it has been fixed in https://github.com/AltBeacon/android-beacon-library/pull/1195. If so, what is the plan for releasing the new version with this fix? 

Thanks in advance!

davidgyoung, 2024-11-16:
closed via #1195

closed 2024-11-16 by davidgyoung