AltBeacon/android-beacon-library #1211: the altBeacon library does not update the Live Data object

G-Roman1988, 2024-11-11:
I have integrated the altBeacon library in my project and it does not update the live data object, it finds an IBeacon device if there are many IBeacon devices nearby, it does not identify them or update the live data. I also noticed this problem in the example application (beaconreferenceapplication) in the library (altbeacon). I noticed that the library depends on a version of other dependencies (libraries). and what libraries and versions are required? where can I read what versions I need for the required dependencies (libraries)? ### Expected behavior
to find each device and update live data.
### Actual behavior
finds an ibeacon device and further finds nothing.
### Steps to reproduce this behavior
integrate altbeacon library into project, start scanning with foreground service and read data from live data or compile examples from library and runazo having multiple ibeacon devices nearby (more than 1).
### Mobile device model and OS version
google pixel 9 pro android 15 and samsung a50 android11
### Android Beacon Library version
V2.20 and with V2.20.6 same result
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2024-11-14:
Regarding this " I also noticed this problem in the example application (beaconreferenceapplication) in the library (altbeacon)"

Did you test with the official Kotlin reference app?  You should see it give a callback to the ranging observer every ~ 1 second here:

https://github.com/davidgyoung/android-beacon-library-reference-kotlin/blob/master/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt#L140

The monitoring observer, however, is expected to get exactly one callback when the beacon appears, and one again when it disappears:

https://github.com/davidgyoung/android-beacon-library-reference-kotlin/blob/master/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.kt#L130

Is this not what you see?

G-Roman1988, 2024-11-16:
yes i tested with the official library app and i get only one iBeacon device. in the diary I see that the monkeys ignore them, I think they are other types of devices, the only difference between my project and the official one is that for me it is not through a simple list but through an adapter. Below is the code from my project.  package org.altbeacon.beaconreference

import android.app.*
import android.content.Context
import android.content.Intent
import android.util.Log
import androidx.core.app.NotificationCompat
import androidx.lifecycle.Observer
import org.altbeacon.beacon.*
import org.altbeacon.bluetooth.BluetoothMedic

class BeaconReferenceApplication: Application() {
    // the region definition is a wildcard that matches all beacons regardless of identifiers.
    // if you only want to detect beacons with a specific UUID, change the id1 paremeter to
    // a UUID like Identifier.parse("2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6")
    private val bleUUID = "A134D0B2-1DA2-1BA7-C94C-E8E00C9F7A2D" //"2D7A9F0C-E0E8-4CC9-A71B-A21DB2D034A1"
    var region = Region("all-beacons", Identifier.parse(bleUUID), null, null)

    override fun onCreate() {
        super.onCreate()

        val beaconManager = BeaconManager.getInstanceForApplication(this)
        BeaconManager.setDebug(true)

        // By default the AndroidBeaconLibrary will only find AltBeacons.  If you wish to make it
        // find a different type of beacon, you must specify the byte layout for that beacon's
        // advertisement with a line like below.  The example shows how to find a beacon with the
        // same byte layout as AltBeacon but with a beaconTypeCode of 0xaabb.  To find the proper
        // layout expression for other beacon types, do a web search for "setBeaconLayout"
        // including the quotes.
        //
        //beaconManager.getBeaconParsers().clear();
        //beaconManager.getBeaconParsers().add(new BeaconParser().
        //        setBeaconLayout("m:0-1=4c00,i:2-24v,p:24-24"));

        // By default the AndroidBeaconLibrary will only find AltBeacons.  If you wish to make it
        // find a different type of beacon like Eddystone or iBeacon, you must specify the byte layout
        // for that beacon's advertisement with a line like below.
        //
        // If you don't care about AltBeacon, you can clear it from the defaults:
        //beaconManager.getBeaconParsers().clear()

        // Uncomment if you want to block the library from updating its distance model database
        //BeaconManager.setDistanceModelUpdateUrl("")

        // The example shows how to find iBeacon.
        val parser = BeaconParser().
        setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24")
        parser.setHardwareAssistManufacturerCodes(arrayOf(0x004c).toIntArray())
        beaconManager.getBeaconParsers().add(
            parser)

        // enabling debugging will send lots of verbose debug information from the library to Logcat
        // this is useful for troubleshooting problmes
        // BeaconManager.setDebug(true)

        // The BluetoothMedic code here, if included, will watch for problems with the bluetooth
        // stack and optionally:
        // - power cycle bluetooth to recover on bluetooth problems
        // - periodically do a proactive scan or transmission to verify the bluetooth stack is OK
        // BluetoothMedic.getInstance().legacyEnablePowerCycleOnFailures(this) // Android 4-12 only
        // BluetoothMedic.getInstance().enablePeriodicTests(this, BluetoothMedic.SCAN_TEST + BluetoothMedic.TRANSMIT_TEST)

        //setupBeaconScanning()
    }
    fun setupBeaconScanning() {
        val beaconManager = BeaconManager.getInstanceForApplication(this)

        // By default, the library will scan in the background every 5 minutes on Android 4-7,
        // which will be limited to scan jobs scheduled every ~15 minutes on Android 8+
[... 261 lines cut ...]
    override fun onCreateViewHolder(parent: ViewGroup, viewType: Int): ViewHolder {
        val itemView = LayoutInflater.from(parent.context).inflate(R.layout.item_device, parent, false)
        return ViewHolder(itemView)
    }

    override fun onBindViewHolder(holder: ViewHolder, position: Int) {
        val device = devices[position]
        holder.bind(device)
    }

    override fun getItemCount() = devices.size
    fun updateDevices (newDevices: List<IBeacon>){
        val previousSize = devices.size
        devices.clear()
        devices.addAll(newDevices)
        if (previousSize <devices.size) {
            notifyItemRangeInserted(previousSize, devices.size - previousSize)
        } else if (previousSize > devices.size) {
            notifyItemRangeRemoved(devices.size, previousSize - devices.size)
        } else {
//            notifyDataSetChanged()
        }
    }

}

davidgyoung, 2024-11-16:
Hey @G-Roman1988, I'd like to help, but this is not the proper forum for troubleshooting a custom app.  

The comment above shows 354 lines of code, which is beyond the scope of what we can do here.  Two suggestions:

1. Can you please make a simpler test case to produce the problem that shows a smaller diff from the reference app (e.g. just a dozen or so lines of code changed?)

2. If you can't do the above but need help troubleshooting a specific use case in a custom app, you may wish to post your question on StackOverflow.com.

G-Roman1988, 2024-11-17:
Hei @davidgyoung Me too please help me today I will open an issue/question on StackOverflow.com. and when it's published I'll leave a link to the StackOverflow.com question here, if you could help me I'd be grateful.

G-Roman1988, 2024-11-21:
Hello @davidgyoungȘi!this is the link to the stackoverflow question  https://stackoverflow.com/questions/79204838/altbeacon-library-not-updating-livedata-with-multiple-ibeacon-devices  . please help me

davidgyoung, 2025-03-07:
Question resolved on StackOveflow.com at the link above

closed 2025-03-07 by davidgyoung