AltBeacon/android-beacon-library #1219: 2 beacons can be scanned via direct android BLE API, but could not be called in beaconManager.addMonitorNotifier

shaojun, 2025-01-23:
question original posted at:
https://stackoverflow.com/questions/25027983/is-this-the-correct-layout-to-detect-ibeacons-with-altbeacons-android-beacon-li/79380414#79380414

I have 2 beacons nearby, and the manufacturer indicates their UUID are:
```
beacon uuid: fda50693-a4e2-4fb1-afcf-c6eb07647825
```
by using the **direct** Android API, I can constantly(every seconds) see the beacon mac and `scanRecord` via:
```
ScanCallback leScanCallback =
    new ScanCallback() {
        @Override
        public void onScanResult(int callbackType, ScanResult result) {
            super.onScanResult(callbackType, result);
            // scanned result data bytes sample:onScanResult-> deviceName: R24110120, rssi: -52, 
            // deviceMacAddress: 52:0A:24:11:00:78, 
            //scanRecord: 0201061aff4c000215fda50693a4e24fb1afcfc6eb0764782500010002d80a0952323431313031323011160318520a24110078000100020603e864000000
        }
};
var bluetoothLeScanner = BluetoothAdapter.getDefaultAdapter().getBluetoothLeScanner();
var bluetoothLeScanner = BluetoothAdapter.getDefaultAdapter().getBluetoothLeScanner();
            ScanSettings settings = new ScanSettings.Builder().build();
            ScanFilter scanFilter = new ScanFilter.Builder().build();
            ArrayList<ScanFilter> scanFilters = new ArrayList<>();
            scanFilters.add(scanFilter);
            bluetoothLeScanner.startScan(scanFilters, settings, leScanCallback);
```

while if I choose the library for scanning:
```
beaconManager = BeaconManager.getInstanceForApplication(this);
// To detect proprietary beacons, you must add a line like below corresponding to yo
// type.  Do a web search for "setBeaconLayout" to get the proper expression.
beaconManager.getBeaconParsers().add(new BeaconParser().
        setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
beaconManager.addMonitorNotifier(new MonitorNotifier() {
    @Override
    public void didEnterRegion(Region region) {
        var regionLogStr = String.format("didEnterRegion: %s", region.toString());
        Log.i(TAG, "onScanResult - I just saw an beacon for the first time! - " + re
    }
    @Override
    public void didExitRegion(Region region) {
        Log.i(TAG, "onScanResult - I no longer see an beacon");
    }
    @Override
    public void didDetermineStateForRegion(int state, Region region) {
        var regionLogStr = String.format("didDetermineStateForRegion: %s, mac: %s", region.toString(), region.getBluetoothAddress());
        Log.i(TAG, "onScanResult - I have just switched from seeing/not seeing beacons: " + state + " - " + regionLogStr);
    }
});
beaconManager.startMonitoring(new Region("myMonitoringUniqueId", null, null, null));
```

I could never have any of that callback: `didEnterRegion` hit, only the `didDetermineStateForRegion` get called when app just started with logging: 
> onScanResult - I have just switched from seeing/not seeing beacons: 0 - didDetermineStateForRegion: id1: null id2: null id3: null, mac: null

what could be wrong here?

davidgyoung, 2025-01-25:
@shaojun please post support questions to StackOverflow.com.  You can include your details above, but please also post why how you know the beacons are actually transmitting.   You may wish to try the BeaconScope app from the Google Play store to confirm the beacons are transmitting, and what format they are using.  

Again, please post a followup to Stackoverflow.com

closed 2025-01-25 by davidgyoung

shaojun, 2025-01-26:
@davidgyoung i've posted it at https://stackoverflow.com/questions/79381310/altbeacon-lib-can-not-detect-beacons
pls help to take a look.