AltBeacon/android-beacon-library #1221: 2.21.0-beta5 should respect the background periods properly

elican-doenyas, 2025-02-12:
Currently, when the library user configures scan periods for background scanning, the library discards the backgroundBetweenScanPeriod, and uses the backgroundScanPeriod for that instead. This is present in 2 places in BeaconManager class, both in the applySettingsChange method.

1.  Line 258

[image]

2. Line 275

[image]

Also, while I am not sure of this, the default value for backgroundBetweenScanPeriod in Settings.kt file, ScanPeriods data class should not be 0.

@davidgyoung  thanks in advance.

elican-doenyas, 2025-03-05:
@davidgyoung any update on this issue? Anything I can do to assist on the resolution of the matter?

davidgyoung, 2025-03-07:
Thanks for this report, @elican-doenyas.  This is a bug and needs to be fixed.  Would you like to submit a Pull Request with this change?  I will gladly merge.

We can also change the default value for the backgroundBetweenScanPeriod in Settings.kt to be 300000 ms (5 minutes) which matches the default elsewhere in the library.

davidgyoung, 2025-03-15:
Fixed in#1221

closed 2025-03-15 by davidgyoung