AltBeacon/android-beacon-library #1239: Service notification on process restart

NaxxoBG, 2025-05-19:
Hi,

I have attached the BeaconManager instance to a singleton. In my app, I have configured to do beacon scanning as a foreground service and have done the relevant configuration. My problem is that if during the scanning I go into my app's permissions and revoke a permission that is needed for the service to run, the permission revocation kills the app process and then starts it up again and this results in the library's scanning foreground service to immediately show the foreground notification. The thing is, I enabled debugging for the manager and checked logcat, and there is no scanning going on, but the notification still appears after the restart. How can I hide it?

I tried cancelling the notification through the notification manager, but to no avail. The only trick that worked is to create a dummy Region and start and stop scanning, right after I get BeaconManager instance using:

`this.beaconManager = BeaconManager.getInstanceForApplication(appContext);`

I would appreciate some insight into what might be going on.

davidgyoung, 2025-05-25:
Can you modify the library reference app and provide instructions to reproduce?  If so, I will take a look.

https://github.com/davidgyoung/android-beacon-library-reference-kotlin

NaxxoBG, 2025-06-02:
> Regarding the foreground service starting up when the app does not have permission to scan, you could simply wrap your beacon init code in something like this:
> 
> if (ActivityCompat.checkSelfPermission(context, Manifest.permission.BLUETOOTH_SCAN) == PackageManager.PERMISSION_GRANTED) { // Start scanning with foreground service here }
> 
> Would this not solve the problem?

Not exactly because the code will not be run. To reproduce the scenario I am describing, you can start the app and scan for beacons using the foreground service. Then, minimize or dismiss the app from the recent apps manager. Then, if you open the app settings for the app and revoke one of the permissions, the app process will be killed and restarted. That's when the notification appears, even though scanning cannot occur due to the permission being revoked.
I believe that behaviour is defined by the BeaconService or some other library-owned class. The notification's appearance could be somehow triggered by the service autobinding upon the application process being restarted when the permission is revoked.

davidgyoung, 2025-06-14:
> Not exactly because the code will not be run. To reproduce the scenario I am describing, you can start the app and scan for beacons using the foreground service. Then, minimize or dismiss the app [...]

If the notification appears, then that means your app has restarted itself -- sometimes Android will decide to restart an app's service after it crashes or is killed.  When Android restarts the service the app's Application class onCreate is called prior to that service restart and any app initialization code is run.  This give you an opportunity to run any code that you want.   I suspect you could add a code block that detects if scanning permissions are not granted, and if so, proactively stop beacon ranging and monitoring.  if you do this, I believe the foreground service will stop, which I think would be an appropriate way to eliminate this problem.