AltBeacon/android-beacon-library #1241: I keep getting the message "Scan failed: app cannot be registered".

bluendev, 2025-06-18:
My app constantly scans for beacons within a given arbitrary layout.

I'm following the basic logic of starting startRangingBeacons if inside and stopRangingBeacons if outside via the **didDetermineStateForRegion** function.

If the scan makes repeated start requests, you can see that additional scan starts are prevented by SCAN_FAILED_ALREADY_STARTED in the callback within the **CycledLeScannerForLolipop** class.

However, if the app is left unused for more than 3 days, the SCAN_FAILED_APPLICATION_REGISTRATION_FAILED message will appear in the callback when starting the scan.

Sometimes the issue can be resolved by rebooting the device or turning Bluetooth off and on, but in some cases it may be temporary or not resolved.

This symptom did not occur at all when targetSdkVersion was 33. Now, I have increased targetSdkVersion to 34 and most (almost all) of the cases experiencing this symptom are on Android OS 15.

1. Is there a way to initialize the scanner space? 
2. Is there a way to find out if the scanner space is not freed because the reference to the scan that did not execute the stop Scan function is gone?

davidgyoung, 2025-06-21:
When you say "unused for more than 3 days" do you mean the app was not running at all for three days and then started itself in the background?  Or do you mean the app was running for 3 days in the background, but not scanning?

Does bringing the app to the foreground make any difference as to whether or not a scan can start?  What about killing the app and restarting?  What about a different app?  Can a different app, say the library reference app or BeaconScope from the Google Play store, successfully start a scan and detect beacons if manually started after you see this condition on your app?

davidgyoung, 2025-06-26:
It sounds like this is a case of a scan client leak.  There may be some case where a scan start contains a client handle which is never released by stopping a scan. And if this repeats itself for long enough all scan clients are exhausted.

Like memory leaks, finding the source of problems like this are always difficult.  I do not think it is realistic to resolve it unless you can reproduce locally.  If we can find the steps to do, you can set up a long running session sending LogCat output to disk with AndroidBeaconLibrary debug logging turned on.  This would allow us to find the source of the problem by finding mismatched scan start and scan stop events in the log.

Have you tried reproducing with a test phone on Android 15 over the course of several days?  You may need to do this with beacons appearing and disappearing periodically before the problem will show up.  I would focus the effort on trying to reproduce, then we can certainly find the problem.

bluendev, 2025-07-02:
I will try to reproduce the phenomenon.

However, in the latest version 2.21.1, when I turn off the screen, the scan stops and starts periodically. Can you tell me where this is happening?

 [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:17.543  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:17.543  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Stop Scan with callback
2025-07-02 10:31:17.547  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:20.546  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:20.546  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  could not find callback wrapper
2025-07-02 10:31:21.546  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:21.547  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  could not find callback wrapper
2025-07-02 10:31:21.547  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:21.548  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Start Scan with callback
2025-07-02 10:31:21.550  4562-4845  BluetoothLeScanner      bluen.bn_smartkey                    D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2025-07-02 10:31:21.554  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:25.544  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:25.544  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Stop Scan with callback
2025-07-02 10:31:25.548  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:28.547  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:28.547  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  could not find callback wrapper
2025-07-02 10:31:29.545  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:29.545  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  could not find callback wrapper
2025-07-02 10:31:29.546  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:29.547  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Start Scan with callback
2025-07-02 10:31:29.549  4562-4845  BluetoothLeScanner      bluen.bn_smartkey                    D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2025-07-02 10:31:29.554  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:33.543  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:33.543  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Stop Scan with callback
2025-07-02 10:31:33.548  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:36.547  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:36.547  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  could not find callback wrapper
2025-07-02 10:31:37.545  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:37.545  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Start Scan with callback
2025-07-02 10:31:37.547  4562-7494  BluetoothLeScanner      bluen.bn_smartkey                    D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2025-07-02 10:31:37.553  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:41.544  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:41.545  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Stop Scan with callback
2025-07-02 10:31:41.549  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0
2025-07-02 10:31:44.550  4562-4872  BluetoothAdapter        bluen.bn_smartkey                    D  isLeEnabled(): ON
2025-07-02 10:31:44.550  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  could not find callback wrapper
[... cut ...]
2025-07-02 10:31:45.546  4562-4872  BluetoothLeScanner      bluen.bn_smartkey                    D  Start Scan with callback
2025-07-02 10:31:45.548  4562-4583  BluetoothLeScanner      bluen.bn_smartkey                    D  onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2025-07-02 10:31:45.553  4151-6400  ScanManager             com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: bluen.bn_smartkey, scannerId: 12, reportDelayMillis=0

davidgyoung, 2026-01-17:
the background scan on/off may be being performed by the operating system -- all the logs lines about it are from the OS. You might turn on debug logging in the library with BeaconManager.setDebug(true) to see if the library logs anything about it

closed 2026-09-09 by bluendev